adafruit/Adafruit_MAX31865 #1: Bias voltage remains on after one-shot read

drhaney, 2017-02-23:
The MAX31865 board dumps a whopping ~4 ma into a Pt100 sensor at room temperature causing visible self-heating while the resistance stabilizes. Sampling it less frequently made no difference until the bias voltage was shut off after reading in ReadRTD(). With the bias voltage strobed only as needed and the RTD sampled at 15+ second intervals, self-heating was at a tolerable minimum.

A short patch file is attached:
[max31865_patch.txt](https://github.com/adafruit/Adafruit_MAX31865/files/796893/max31865_patch.txt)

drhaney, 2018-03-19:
Github sucks and I hate it.
You do it.
It is essentially a 1 line addition into the driver library of a deeply flawed chip.

tannewt, 2018-03-19:
@drhaney Please relax. Complaining doesn't help anyone.

I'll take a look at the patch soon. Thanks for providing it.

drhaney, 2018-03-19:
[The essential experience of Git and Github](https://xkcd.com/1597/)

xmenxwk, 2018-03-22:
>  deeply flawed chip.

how ?

drhaney, 2018-03-22:
Re: MAX31865
>> deeply flawed chip.
>
> how ?

See original post above.

That and more issues are discussed on the Adafruit forum thread
[MAX31865 correction and concerns](https://forums.adafruit.com/viewtopic.php?f=8&t=112346&p=561615&hilit=max31865#p561615)

Maxim's own [reference design board for RTD sensors](https://www.maximintegrated.com/en/design/reference-design-center/system-board/6185.html/) runs them at 0.8ma which demonstrates that they are aware of the issue.  And it _**doesn't**_ use their own MAX31865.

If you want accurate temperature readings from a Pt100 RTD, you DON'T use the MAX31865.

Search on "**Pt100 RTD self-heating**" for abundant overviews of the problem.

xmenxwk, 2018-03-22:
so you are saying, this chip has accuracy of +-1 degree because it has 0.1% reference resistor ? and also heats up the sensor by allowing more current to flow through it ?

drhaney, 2018-03-22:
You need to run the numbers and draw your own conclusions.  The Pt100 board needs a two point temperature calibration to overcome the combination of self-heating, sense resistor tolerance, and RTD error margin.  Without that, you may as well use a thermocouple.  The TI ADS1118 and a K thermocouple do well even uncalibrated.

xmenxwk, 2018-03-22:
have you ever tested this ?
https://github.com/drhaney/pt100rtd

drhaney, 2018-03-22:
RE: pt100rtd library

I wrote it.

drhaney, 2018-03-22:
>> have you ever tested this ?
>https://github.com/drhaney/pt100rtd

It was downloaded 49 times from the Adafruit forum, garnered 5 stars on Github, with neither complaints nor corrections yet so  I suspect it has been adequately tested.

emmanuel-nwogu, 2020-09-17:
@drhaney  Does your library solve the self-heating problem? If it doesn't, will turning off bias voltage after reading, delaying, and turning it back on for the next reading help? I'm working on a project that needs to take readings periodically. Would appreciate the input.

emmanuel-nwogu, 2020-09-17:
I dug into the adafruit max31865 library and found out the readRTD() enables bias voltage before reading and disables it after. What made that unclear at first was that the disabling did not use the enableBias() method.

caternuson, 2020-09-30:
Fixed with #30

closed 2020-09-30 by caternuson