adafruit/Adafruit_MAX31865 #4: 0x40 fault

puhycz, 2017-09-19:
I try demo with Arduino Nano and it´s print only 0x40 fault. 

RTD value: 0
Ratio = 0.00000000
Resistance = 0.00000000
Temperature = -242.02
Fault 0x40
RTD Low Threshold

Please what it s mean?

Thanks,

Jan

ladyada, 2017-09-19:
probably a wiring issue. if you purchased from adafruit you can post in our forums for tech support! :)

closed 2017-09-19 by ladyada

SanjayChakrapani, 2020-10-02:
If you are using 3wired PT100, 
1.make sure that you cut 2-way jumper pad 
2.short the right-side pads together
3.also short 2/3 wire pad
4. **thermo.begin(MAX31865_3WIRE);**
[image]
https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/rtd-wiring-config#3-wire-sensors

SanjayChakrapani, 2020-10-02:
> 
> 
> I try demo with Arduino Nano and it´s print only 0x40 fault.
> 
> RTD value: 0
> Ratio = 0.00000000
> Resistance = 0.00000000
> Temperature = -242.02
> Fault 0x40
> RTD Low Threshold
> 
> Please what it s mean?
> 
> Thanks,
> 
> Jan

If you are using 3wired PT100, 
1.make sure that you cut 2-way jumper pad 
2.short the right-side pads together
3.also short 2/3 wire pad
4. **thermo.begin(MAX31865_3WIRE);**
[image]
https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/rtd-wiring-config#3-wire-sensors