adafruit/Adafruit_MAX31865 #10: Two delays in readRTD are blocking

MakeNModify, 2019-05-31:
In Adafruit_MAX31865::readRTD two delays are used. This makes it unusable when driving a stepper motor at the same time (motor stops at every reading). Is there an easy way around it?

The way it is it is hard to be used for simultaneous tasks, it would be nice if this could be changed.
Thank you :)

ladyada, 2019-05-31:
at this time, no - you can use interrupts for other tasks or you could submit a PR for non-blocking reads, but we have no plans to do so.

MakeNModify, 2019-06-02:
The project is somewhat like an 3D Printer Extruder/Hotend. So the Stepper has a high priority. If so it would work better to stop measuring temperature while extruding (3sec). But that is an compromise I'm not willing to make. Sure i could transfer the temperature control task to a second arduino but that would be solving software issues with hardware which is most of the time not an efficient concept. 

Anyways thank you for your fast feedback. Meanwhile I found an other library that is sufficient for what I need and dose not block here:
https://github.com/hallard/arduino-max31865

rvt, 2020-08-09:
The library could be easily be written such that we can do both.
Keep the readRTD function with the delay.
And for people like me andMakeNModify two additional function that are called ´readRTDStart(..)' and 'readRTDread(...)' (open for better names)...
Then people can decide if they want the 'async' or 'sync' version.
This would allow for code re-use.
Interrupts are not to be used for ´tasks´ If you are open for a PR around this, I can get you one.

sylvanoMTL, 2021-02-16:
[Here are some modifications](https://forum.arduino.cc/index.php?topic=703346.msg4772457#msg4772457), sample code is provided [here](https://forum.arduino.cc/index.php?topic=703346.msg4898400#msg4898400)  to read multiple RTDs with a single shot. 

Here are some results: 

1RTD single   shot | 76 to 77 ms
8 RTDs single   shot | 614 to 616   ms    (77ms * 8 = 616   ms)
15 RTDs   single shot | 1152 to 1153   ms  (77ms*15 = 1555 ms)
8 RTDs single   shot but in asynchronous mode | 76 to 77ms
15 RTDs   single shot but in asynchronous mode | 76 to 77ms

It would be nice to see a modification of the library to have also continuous measurement. I am sure that such modification combined with asynchronous reading would help to reach the 20ms reading.

frowin, 2021-05-09:
I modified the library so that it becomes async. Until now it's a little bit hacky (e.g. it returns a temperature of 0 when the sensor is not yet ready to be read out). For timing I used millis() instead of delay(). With this change, the loop is not blocked for 75ms per sensor but only 160us.

rvt, 2021-05-09:
If possible, I would let the temperature return NaN if it's unknown, and?or at least a function you can call to know if a temperature has a valid reading or not.

sylvanoMTL, 2021-05-09:
After forking some codes, and taken some piece of code from the link I posted above, I am using this [forked library](https://github.com/sylvanoMTL/Adafruit_MAX31865). it could be clean up and pulled but I have very little time  for this. As well I have an example on how to use this forked library with the following options:
1) One-shot (typical example)
2) Asynchro one-shot
3) Continuous
I will add it if I have some time.

sylvanoMTL, 2021-05-09:
I think you may be right, but for me it works well
The state machine is in reference with the property :  uint32_t chrono;  line 125 of the .h file. 
I believe the variable "state" stay internal to the function.
```
void setup(){
Adafruit_Max31865 RTD1;
Adafruit_Max31865 RTD2;
...
}

void loop(){
 RTD1.readRTDAsynch(); //will create an internal state
 RTD2.readRTDAsynch(); //will create another internal state (tbc)

```
   
This is to be confirmed, I had no issue plugin up to 15 RTDs on one arduino micro.

sylvanoMTL, 2021-05-09:
Thanks, I would need to pass that variable as a class attribute.

sylvanoMTL, 2021-07-24:
I have completed a first correction for the asynchronous mode: https://github.com/sylvanoMTL/Adafruit_MAX31865/tree/asynchronous_mode_fix
I haven`t written an example yet, but my this library works fine with my project. as discuss above, `state` and `chrono` is an attribute of the class, and are modified by the method `setState(t_state new_state)`. 
this method is called in  `RTD.begin( );` and `readRTDAsync(uint16_t& rtd);`

Work to do is to clean up comments in the .h and .cpp and provide a .ino for the general user. I will do that when I get more time in my hands.

studi1882, 2022-05-30:
Thank you for your work! With this modification it also works parallel which is pretty amazing! 10 Pt100 in 76 ms instead of 760 ms!

budulinek, 2023-10-04:
For a non-blocking version of the library, you can also check my fork:

https://github.com/budulinek/MAX31865_NonBlocking

A state machine is not implemented in the library itself. But you have one in examples (check examples/MAX31865_Single ).