adafruit/Adafruit_MAX31865 #21: Issue with version 1.2.0

soothsayer011, 2020-05-22:
This library is used in Marlin 2.0 firmware for 3D printing, and since the changes made to Adafruit_MAX31865 yesterday from 1.1.0 to 1.2.0, the firmware no longer compiles. Here is a thread on the issue https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/257

The firmware will compile with version 1.1.0.

Here is the output of the error which references the library
ompiling .pio\build\STM32F103RC_btt_512K\liba17\USBComposite\Consumer.cpp.o
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
^
In file included from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
uint8 requestFrom(uint8, int);
^~~~~~~~~~~
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate expects 2 arguments, 3 provided
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
uint8 requestFrom(int, int);
^~~~~~~~~~~
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate expects 2 arguments, 3 provided
*** [.pio\build\STM32F103RC_btt_512K\lib2ab\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:55:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:0:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' [-Wconversion-null]
Adafruit_SPIDevice spi_dev = NULL;
^~~~
In file included from c:\users\lastp.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\inttypes.h:22:0,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:4,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:50,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' uses the default argument for parameter 3, which is not yet defined
Adafruit_SPIDevice spi_dev = NULL;
^
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t, int8_t, int8_t, int8_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:38:76: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BitOrder, uint8_t)' uses the default argument for parameter 6, which is not yet defined
spi_dev = Adafruit_SPIDevice(spi_cs, spi_clk, spi_miso, spi_mosi, 1000000);
^
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\lib1cd\Adafruit_MAX31865\Adafruit_MAX31865.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:2:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
[... 27 lines cut ...]
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
_spi->transfer(buffer, len);
^
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
uint8_t transfer(uint8_t data) const;
^~~~~~~~
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate expects 1 argument, 2 provided
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: note: suggested alternative: 'SPI_CR1_LSBFIRST'
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:149:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:149:23: note: suggested alternative: 'SPI_CR1_LSBFIRST'
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\lib2ab\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1

ladyada, 2020-05-22:
looks like Maple has a completely different way of doing I2C and SPI. for now we recommend pinning to the previous version. the chip you are using is "STM32F103"??

soothsayer011, 2020-05-22:
Yes, STM32F103RC