adafruit/Adafruit_MAX31865 #22: Daisy chain mode

edouardreg, 2020-05-28:
Hello,

As temperatures measurements often involves several sensors, it would be great if there was a daisy chain possibility with this library to limit the numbers of GPIO used when you have 2,3,...8 max 31865 in a project.
Could it be possible to write a "daisy chain version" of the function :
'''float Adafruit_MAX31865::temperature(float RTDnominal, float refResistor)'''

Or explain how use adafruit max31865 module in daisy chain mode if already possible with this library ?

Thanks !

caternuson, 2020-09-30:
Are you requesting support for this configuration?
https://www.maximintegrated.com/en/design/technical-documents/app-notes/3/3947.html
Not sure if the MAX31865 supports that.

For the more common was of doing this - you can have more than one MAX31865 as long you have a separate CS for each one.

caternuson, 2022-06-21:
Closing due to lack of response. Need more information on what "daisy chain mode" is.

closed 2022-06-21 by caternuson