adafruit/Adafruit_MAX31865 #23: error when going to 1.2.0 from 1.1.0

janbbeck, 2020-05-31:
simply changing the version from `1.1.0` to `1.2.0` in the library manager causes the following error to appear for me:
```fatal error: Adafruit_SPIDevice.h: No such file or directory```

- Arduino board:  ESP32 TTGO T-Display

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.8

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Sorry, I can't share the sketch. This happens on recompile. It's consistent - switch to 1.1.0, and it works, switch to 1.2.0 and it breaks. My guess is that it forgets to download the .h file for the newer lib.

caternuson, 2020-05-31:
Try installing the BusIO library:
https://github.com/adafruit/Adafruit_BusIO
which should be available via Library Manager.

That's a new requirement for the 1.2.0 release. Maybe the upgrade process isn't bringing it in for some reason?

janbbeck, 2020-05-31:
Confirmed. Manual install of BusIO removes the issue.

caternuson, 2020-05-31:
Cool. Glad that worked.

closed 2020-05-31 by caternuson

ladyada, 2020-05-31:
yeah 1.8.8 is a bit old now, dep management showed up in 1.8.10 - we recommend upgrading :)