adafruit/Adafruit_MAX31865 #26: Adafruit_MAX31865 library does not work with software SPI

tho1010, 2020-08-29:
MAX31865 library with software SPI does not work properly and false data is generated with the example sketch max31865.ino

Work around: 
1) Use hardware SPI only
2) Replace the read and write functions of the MAX31865 class library by using the corresponding calls to SPI.h

minor issue: 
after Serial.begin(115200); a delay of approx. 1000 should be added to allow for settling time of the serial port connection.

ladyada, 2020-08-29:
try changing this line

https://github.com/adafruit/Adafruit_MAX31865/blob/master/Adafruit_MAX31865.cpp#L38

to

`spi_dev = Adafruit_SPIDevice(spi_cs, spi_clk, spi_miso, spi_mosi, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);`

tho1010, 2020-08-29:
Perfect! 
The additional SPI_MODE Parameter does the trick! 
Software SPI works now with both SPI_MODE1 and 3 as defined in the MAX31865 datasheet

closed 2020-08-29 by tho1010