adafruit/Adafruit_MAX31865 #27: Adafruit_MAX31865::Begin does not initiate the "Fault Detection Cycle"

tho1010, 2020-09-29:
According to the MAX31865 data sheet (page 14) a manual or automatic fault detection cycle must be run at startup.
Without that the startup of the MAX31865 behaves arbitrarily and can prevent data reading, which in occurred to me at every second boot cycle.

I solved this issue by adding the following code at the beginning of my local copy of Adafruit_MAX31865::Begin to enable "Automatic fault detection":

#define MAX31865_CONFIG_AUTO_FAULT_DETECT 0b10010101
  uint8_t t = readRegister8(MAX31856_CONFIG_REG);
  t  |= MAX31865_CONFIG_AUTO_FAULT_DETECT;
  writeRegister8(MAX31856_CONFIG_REG, t);

I  recommend to extend the begin function accordingly by defining a new "Fault detection" function, ideally configurable for automatic or manual fault detection.

[MAX31865.pdf](https://github.com/adafruit/Adafruit_MAX31865/files/5299165/MAX31865.pdf)

ladyada, 2020-09-29:
OK! - wanna add it and send in a PR ? :)

AnHardt, 2020-09-30:
Related to #5 .

caternuson, 2022-06-21:
> According to the MAX31865 data sheet (page 14) a manual or automatic fault detection cycle must be run at startup.
> Without that the startup of the MAX31865 behaves arbitrarily and can prevent data reading, 

Where is this in the datasheet? Read page 14, but not seeing where it indicates this behavior. Writing to the config register is needed initiate a manual or automatic fault detection cycle. But not seeing where it indicates the cycle must be run at startup.