adafruit/Adafruit_MAX31865 #34: hardware SPI asumes it is always HSPI and does not allow to use VSPI

pabloandresm, 2021-12-09:
Even though Adafruit_SPIDevice library allows to pass the hardward SPI as parameter in its constructor, it is not reflected in this library's constructor.

So Adafruit_MAX31865 hardware SPI assumes Adafruit_SPIDevice with SPIClass = SPI, which is defined in the SPI.h as HSPI (meaning pins 12, 13, 14, 15).

The Adafruit_SPIDevice  allows to select the hardware SPI VSPI (pins 19, 18, 5, 23), but this library doesn't.

The constructor must allow to pass a hardware SPI.

The ESP32 has 3 hardware SPIs: SPI, HSPI, and VSPI

ladyada, 2021-12-09:
yah this was written a while ago when ESP32 didnt exist - please submit a PR to add a constructor argument for optional SPI device!

caternuson, 2021-12-09:
I intentionally left this out of the BusIO update. Didn't want to do too many things in one PR. Should be an easy update.

caternuson, 2022-06-21:
@pabloandresm Can you test #34 on your ESP32 setup and try the alternate hardware SPI ports?

caternuson, 2022-06-29:
Closing. Added with #38

closed 2022-06-29 by caternuson